ma1co/OpenMemories-Tweak #411: Ma1co

Cuberlain, 2021-07-10:
Hello!
Why don't you make some little hacks about higher Bitrate in some cameras? It can not be so complicated. Forexample video cameras with 100 Mbit AVC can shot with 120 or 150 Mbit bitrate etc...

dreamsindrums, 2021-07-20:
Have a go yourself if it is not so complicated :)

mungewell, 2021-07-23:
Most of the 'hacking' with this project relies on the features already built into the Sony FW.

You may be able to access higher/different bitrates than your camera's menu offers by changing the following keys.

```
AVCHD bitrate
 id=0x010700a1, size=0x0001, attr=0x60:
4 = 17mbit/s
5 = 24mbi/s
6 = 28mbit/s
```

```
MP4 bitrate
id=0x010700a2, size=0x0001, attr=0x60:
1 = 3Mbit/6MBit
2 = 12Mbit
```

I suspect that there's key for XAVCS as well, but I haven't located it yet.

Cuberlain, 2021-07-25:
Hello Mr. Mungwell!

I have an AX53 and an AX700 camcorder.

If you can do it in 4K modes, it would be really fantastic. Just imagine the Ax700, It has so good quality if you record it via HDMI via HDMI=>USB3 capture stick (Camlink 4K) connected to laptop , using Virtualdub and lossless UTvideo encoder....
So the AX700 deserve more than the 100mbit/s, because 100 Mbit can not capture the details , when you using the camcorder during walking on streets. 
Do we have hop to do the maximal 150Mbit/s in the case of ax53 or AX700? Do you think that both camcorder understand similar codes, because they have the same BIONZ X chipsets. 

Thank you for your reply!
Best wishes!
Bye!

Cuberlain, 2021-07-25:
Hello Mr. Mungewell!

I have an AX53 and an AX700 camcorder.

If you can do it in 4K modes, it would be really fantastic. Just imagine the Ax700, It has so good quality if you record it via HDMI via HDMI=>USB3 capture stick (Camlink 4K) connected to laptop , using Virtualdub and lossless UTvideo encoder....
So the AX700 deserve more than the 100mbit/s, because 100 Mbit can not capture the details , when you using the camcorder during walking on streets. 
Do we have hop to do the maximal 150Mbit/s in the case of ax53 or AX700? Do you think that both camcorder understand similar codes, because they have the same BIONZ X chipsets. 

Thank you for your reply!
Best wishes!
Bye!
> Most of the 'hacking' with this project relies on the features already built into the Sony FW.
>
> You may be able to access higher/different bitrates than your camera's menu offers by changing the [...]

Cuberlain, 2021-07-25:
XAVC maybe based on the aforementioned "MP4 Bitrate?" BEcause 4K is alwayss captured in MP4 container.

mungewell, 2021-07-26:
The MP4 Bitrate is for the MP4 codec (setting 4), my HX60 offers the 2 bitrates (3Mbit/12Mbit).

Codec is set with
```
id=0x01070087, size=0x0001, attr=0x68:
1 = DVD Movie?
2 = M2PS ?
3 = AVCHD
4 = MP4
5 = Still
6 = BGM?
7 = DVD?
a = XAVC S
```

My as100 also only does 50MBit for both it's modes, so I was not able to track down the bitrate control.

If someone has a device which does multiple bitrates for different resolutions they can set each in turn, dump the `Backup.bin`, decode it to text and look for which keys change.

Cuberlain, 2021-07-26:
Hello!

The  AX100, the AX53 and AX700 have XAVC s modes in 60Mbit/s and 100mbit/s in their bitrate selector.
Where can I found that Backup.BIN in the SD CARD??
I see only status.bin in the  :\PRIVATE\M4ROOT Folder.

Will it help you if you downoad the firmware update files of AX53 or AX700 cameras?

Cuberlain, 2021-07-26:
Hello!

The AX100, the AX53 and AX700 have XAVC s modes in 60Mbit/s and 100mbit/s in their bitrate selector in 4K resolution.
Where can I found that Backup.BIN in the SD CARD??
I see only status.bin in the :\PRIVATE\M4ROOT Folder.

Will it help you if you downoad the firmware update files of AX53 or AX700 cameras?

> 
> 
> The MP4 Bitrate is for the MP4 codec (setting 4), my HX60 offers the 2 bitrates (3Mbit/12Mbit).
> 
> Codec is set with
> 
> ```
> id=0x01070087, size=0x0001, attr=0x68:
> 1 = DVD Movie?
> 2 = M2PS ?
> 3 = AVCHD
> 4 = MP4
> 5 = Still
> 6 = BGM?
> 7 = DVD?
> a = XAVC S
> ```
> 
> My as100 also only does 50MBit for both it's XAVCS modes, so I was not able to track down the bitrate control.
> 
> If someone has a device which does multiple bitrates for different resolutions they can set each in turn, dump the `Backup.bin`, decode it to text and look for which keys change.

mungewell, 2021-07-26:
So again, before too many hopes are elevated - most of the 'hacking' here is to enable/control functionality **already** built into the camera(s). For example functionality not exposed by the on-camera menu system. There may be good reason why the menu does not allow, for example it might cause the camera to overheat or crash frequently.

Assuming that your camera is supported by the tools... the `Backup.bin` is not normally accessible you have to access the 'internals' of the camera.

My technique for reading the keys en-mass, is to use `pmca-console` in it's `updatershell` mode. This lets you do `pull /setting/Backup.bin 1080p60_60mbit.bit` to download a local copy with a particular video setting, then disconnect (`exit`) from camera, pull USB cable, change the video setting on the camera's menu and repeat the download process.

Then I use `python3 fwtool.py print_backup -f 1080p60_60mbit.bin > 1080p60_60mbit.txt` on each of the backup files I downloaded, which expands the settings into a text report of all the keys/values.

Then I look at the txt files and look for differences, this is particular easy with a tool like `meld`. For a small change like changing resolution, or video bitrate I'd only expect to see 1 or 2 keys changing in small ways. _You will see big keys changing, these are not the keys and can be ignored._

https://github.com/ma1co/Sony-PMCA-RE
https://github.com/ma1co/fwtool.py
https://meldmerge.org/

Cuberlain, 2021-07-26:
HI Mungewell!
Does these softwares good for FDR-AX700 Camera too?
Isn't the ISP (in this case BIONZ X) contain the HW encoder too?

Cuberlain, 2021-07-26:
HI Mungewell!
Does these softwares good for FDR-AX700 Camera too?
Doesn't the ISP (in this case BIONZ X chipset) contain the HW encoder too?

> So again, before too many hopes are elevated - most of the 'hacking' here is to enable/control functionality **already** built into the camera(s). For example functionality not exposed by the [...]

Wirtigl, 2022-02-05:
> So again, before too many hopes are elevated - most of the 'hacking' here is to enable/control functionality **already** built into the camera(s). For example functionality not exposed by the [...]

Hello Mungewell!

I opened a new topic for that. Can you help to solve this problem?
https://github.com/ma1co/Sony-PMCA-RE/issues/284