ma1co/OpenMemories-Tweak #423: Clean HDMI out for NEX6?

domsob1974, 2021-09-16:
Is it possible to get clean HDMI for the NEX 6 using a hidden setting?  It seems as if thousands of people online have been asking for this and, if possible, it would certainly breathe new life into these old cameras.

kaixxx, 2021-12-28:
Has been discussed here: https://github.com/ma1co/Sony-PMCA-RE/issues/249
No solution yet, unfortunately.

domsob1974, 2022-10-28:
Any progress on getting clean HDMI out from the Nex6?  I'd donate for that to happen :)

GreyAsteroid, 2023-10-21:
I can verify that the solution posted [here](https://github.com/ma1co/Sony-PMCA-RE/issues/448) works for my NEX-5R. I would assume it also works for the NEX-6 but I don't believe it has been tested yet.