ma1co/OpenMemories-Tweak #426: A7Rii Constant black Screen

neopixel2, 2021-11-16:
Hi,

Out of nowhere and after 4 years without any issues my A7rii doesn't want to boot up normally anymore.
After searching this thread I realized the culprit was the tweak installed...

I have read and tried most of the suggested things in here but still with no luck. My camera now only wants to turn on while connecting via USB to my laptop. It is very frustrating... Nothing happens when it isn't connect to the laptop. Just the black screen...

If someone ever solved successfully this I would love to hear your story!

Here is what I tried so far:

- Regained access to the camera via usb and disabled all the settings in the tweak app. (I was v.0.10) 
- reinstalled the tweak app and update it to v0.12
- initialized the camera 

So far nothing is happening. the camera still needs to be connected via USB in order to properly boot up.

- dowloaded and installed latest v0.17.
- open tweak app, it says in its meny it is tweak app v0.11
- tried a couple more times, still says on camera its v0.11

Camera still needs USB to boot up.

Any help would be truly appreciated!

Many thanks!

dsol1980, 2021-11-16:
Are we sure it's the tweak? I have the a7r2 as well and The tweak has been installed for years.

neopixel2, 2021-11-16:
Pretty sure yes. The tweak were installed on mine for years as well. Then suddenly it is impossible to start the camera. 

Today I was lucky through USB to regain control. It's been weeks that I was just prepared of getting rid of this body.

I tried to update to v0.17. nothing. I also tried v0.10 v0.11 and v.12. 

The camera was fine so I let it charge the battery through the USB cable.

As I am writing this the battery is probably 100%. But the camera doesn't boot up again. Black screen, no LED lights, nothing. Tried to remove the battery and SD card, turn off/on, replug to USB...nothing...

I am now 100% sure it is software and not hardware.

I would love to just "flash" the body system to go back to it's initial factory settings/code but I find nothing online that would help with that except a $500 Sony repair that I desperately try to avoid...

Also, I tried to update the camera firmware to 4.01. Nothing...

Thanks again for the response, it is truly appreciated