ma1co/OpenMemories-Tweak #427: "Turn Power OFF then ON again" issue

totolight, 2021-11-21:
Hi,

My SONY RX100 M3 boots with the error "E:62:20", then retracts the lens, and then tells me "Turn Power off then ON".
Either the camera goes to the trash, either I find a way to force this message to go away so I can use the camera
I was wondering if there would be a way, via telnet connection perhaps, using the memorytweak app, to make this issue go away.

Many thanks in advance !

gree303, 2022-03-11:
Hi totolight, have you found a solution to this problem? Thanks.

totolight, 2022-03-12:
Hi !
No unfortunately, the camera is still on my table and still behaving the same. If you find something yourself I would be glad if you could share.
You have the same issue ?

KiwiBird97, 2022-03-20:
Try the following:
- Turn off the camera.
- Remove the battery pack.
- Wait for one minute, then reattach the battery pack.
- Turn on the power of the camera.

If that doesn’t work, contact Sony for service.

Source: https://www.sony.com/electronics/support/articles/00008857