ma1co/OpenMemories-Tweak #428: Support to ZV-E10

dexterobj, 2021-12-29:
Could I use this app to Unlock the language menu on a Japanese ZV-E10?

dishmaeyl, 2022-03-23:
Having the same issue as well

benn333, 2022-04-07:
Also in need of support for ZX-E10.

ma1co, 2022-04-30:
In the latest [development build](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts), tweaks should now be supported for this camera, too. You can use `pmca-console serviceshell` or the "Start tweaking (service mode)" button in the Tweaks tab of pmca-gui. This will require the installation of custom USB [drivers](https://github.com/ma1co/Sony-PMCA-RE#windows-drivers).

closed 2022-04-30 by ma1co

bigorbi, 2022-10-23:
Hello everyone, even after installation of custom USB drivers I can`t use Tweaks tab of pmca-gui.. does anyone has the same issue?