ma1co/OpenMemories-Tweak #431: sony a1 a7c a7m3 a7r3 a7m4 a7r4 fx3 ZV-E10 ZV1 A6100 A6400 A6600 RX100M6 RX100M7 Add English menu

Camerasupport, 2022-03-18:
This is a legitimate service You can find me on Taobao.com. Emails can also be sent.

🌐https://m.tb.cn/h.fLELc61?tk=Vnz82gbc8JM

📧wanghaixian1995@gmail.com

💻Remotely, you need to prepare:  
computer (Windows✅ Macos❎)
  Install remote software (TeamViewer)  
  USB cable high speed internet

⬇️⬇️You can also add other languages⬇️⬇️  日本語🇯🇵 English🇺🇸 Français🇫🇷 Deutsch🇩🇪 Español🇪🇸 Italiano🇮🇹 Português🇵🇹 简体中文🇨🇳 繁體中文🇭🇰 العربية🇸🇦 한국어🇰🇷 Nederlands🇳🇱 Русский🇷🇺 ภาษา ไทย🇹🇭 فارسی🇮🇷 Magyar🇭🇺 Česky🇨🇿 Polski🇵🇱 Svenska🇸🇪 Norsk🇳🇴 Türkçe🇹🇷 Ελληνικά🇬🇷 Română🇷🇴 Bahasa Melayu🇲🇾 Indonesia🇵🇱 Suomi🇫🇮 hrvatski🇭🇷 Dansk🇩🇰 Български🇧🇬 Tiếng Việt🇻🇳 Українська🇺🇦 Português (Brasil)🇧🇷 France(Canada) Язык🇨🇦

Lee0078, 2022-03-20:
a6400相机可以刷了吗？

dishmaeyl, 2022-03-23:
I sent you an email

nitinbarve, 2022-04-05:
Thanks to wanghaixian1995@gmail.com
With his help i was able to enable SONY RX100M7 all language settings. :)

ma1co, 2022-04-30:
In the latest [development build](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts), tweaks should now be supported for these cameras, too. You can use `pmca-console serviceshell` or the "Start tweaking (service mode)" button in the Tweaks tab of pmca-gui. This will require the installation of custom USB [drivers](https://github.com/ma1co/Sony-PMCA-RE#windows-drivers).

closed 2022-04-30 by ma1co