ma1co/OpenMemories-Tweak #435: Rx100m5a rx100VA

Riaandw79, 2022-04-24:
Is there any way to force install open memories tweak on the rx 100 m5A? Install reports camera not supported. Probably the same firmware as m6. Would this be included in future development?

The tweak made the m3 bearable, but now stuck without intervalometer and auto-sync to cloud

ma1co, 2022-04-30:
In the latest [development build](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts), tweaks should now be supported for this camera, too. You can use `pmca-console serviceshell` or the "Start tweaking (service mode)" button in the Tweaks tab of pmca-gui. This will require the installation of custom USB [drivers](https://github.com/ma1co/Sony-PMCA-RE#windows-drivers).

If you manage to run it, I would be interested in a firmware dump. You can dump the firmware using `pull /dev/nflashaB0` and `pull /dev/nflasha` in serviceshell. Please upload the files to my [dropbox](https://www.dropbox.com/request/WXQ1IKIoPo4ECK8CFxRx). Thanks!

Riaandw79, 2022-04-30:
Awesome ma1co! Let me see what I can accomplish there. Update to follow.

Riaandw79, 2022-04-30:
Loading app list
Found 11 apps

Using drivers Windows-MSC, Windows-MTP, Windows-vendor-specific, libusb-MSC, libusb-MTP, libusb-vendor-specific
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Model:              DSC-RX100M5A
Product code:       0080817713
Serial number:      05293449
Firmware version:   1.00

Using drivers Windows-MSC, Windows-MTP, Windows-vendor-specific, libusb-MSC, libusb-MTP, libusb-vendor-specific
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Switching to service mode

Waiting for camera to switch...
Found a camera in service mode
Error: Only libusb drivers are supported for service mode

Using drivers Windows-MSC, Windows-MTP, Windows-vendor-specific, libusb-MSC, libusb-MTP, libusb-vendor-specific
Looking for Sony devices
Found a camera in service mode

Authenticating
Done

Using drivers Windows-MSC, Windows-MTP, Windows-vendor-specific, libusb-MSC, libusb-MTP, libusb-vendor-specific
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Switching to app install mode
Error: This device does not support apps

Using drivers Windows-MSC, Windows-MTP, Windows-vendor-specific, libusb-MSC, libusb-MTP, libusb-vendor-specific
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Switching to app install mode
Error: This device does not support apps

Using drivers Windows-MSC, Windows-MTP, Windows-vendor-specific, libusb-MSC, libusb-MTP, libusb-vendor-specific
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Switching to service mode

Waiting for camera to switch...
Found a camera in service mode
Authenticating
Done

Using drivers Windows-MSC, Windows-MTP, Windows-vendor-specific, libusb-MSC, libusb-MTP, libusb-vendor-specific
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Switching to service mode

Waiting for camera to switch...
Found a camera in service mode
Authenticating
Done

Using drivers Windows-MSC, Windows-MTP, Windows-vendor-specific, libusb-MSC, libusb-MTP, libusb-vendor-specific
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Model:              DSC-RX100M5A
Product code:       0080817713
Serial number:      05293449
Firmware version:   1.00

Using drivers Windows-MSC, Windows-MTP, Windows-vendor-specific, libusb-MSC, libusb-MTP, libusb-vendor-specific
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Switching to app install mode
Error: This device does not support apps
 
 
 ....
 
 Not sure if I am doing this correctly.

Riaandw79, 2022-04-30:
When running pmca-console-v0.18-6-g295cbc8-win a cmd window opens and then auto closes. Any other way to dump the firmware?

ma1co, 2022-04-30:
The RX100M5A does not have the android subsystem, so app installation won't work, sadly. However, things like the language tweak should work.

To dump the firmware, pmca-console has to be used. Open a windws command prompt, change to the directory where you have downloaded pmca-console, then run `pmca-console-v0.18-6-g295cbc8-win serviceshell`. Then you can run the two `pull` commands.

Riaandw79, 2022-04-30:
Ah, makes sense now. I found android overlay app for remote control which solves timelapse luckily.

Ok, console working.

log as follows.

C:\>pmca-console-v0.18-6-g295cbc8-win serviceshell
Using drivers Windows-MSC, Windows-MTP, Windows-vendor-specific, libusb-MSC, libusb-MTP, libusb-vendor-specific
Looking for Sony devices
No devices found. Ensure your camera is connected.

C:\>pmca-console-v0.18-6-g295cbc8-win serviceshell
Using drivers Windows-MSC, Windows-MTP, Windows-vendor-specific, libusb-MSC, libusb-MTP, libusb-vendor-specific
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Switching to service mode

Waiting for camera to switch...
Found a camera in service mode
Authenticating
Starting service shell...

Welcome to platform shell.
Type `help` for the list of supported commands.
Type `exit` to quit.
>pull /dev/nflashaB0
Error: [Errno 13] Permission denied: '.\\nflashaB0'
>pull /dev/nflashaB0
Error: [Errno 13] Permission denied: '.\\nflashaB0'
>pull /dev/nflasha
Error: [Errno 13] Permission denied: '.\\nflasha'
>

ma1co, 2022-04-30:
Looks like pmca-console doesn't have write permissions in the current folder on your computer? You can try `pull /dev/nflasha C:\path\to\some\writable\directory`

Riaandw79, 2022-04-30:
That solved it. Uploading now!

C:\Users\dell\Downloads>pmca-console-v0.18-6-g295cbc8-win serviceshell
Using drivers Windows-MSC, Windows-MTP, Windows-vendor-specific, libusb-MSC, libusb-MTP, libusb-vendor-specific
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Switching to service mode

Waiting for camera to switch...
Found a camera in service mode
Authenticating
Starting service shell...

Welcome to platform shell.
Type `help` for the list of supported commands.
Type `exit` to quit.
>pull /dev/nflashaB0
Writing to .\nflashaB0...
>pull /dev/nflasha
Writing to .\nflasha...

Riaandw79, 2022-04-30:
Reports your dropbox is out of storage space...

https://www.dropbox.com/sh/wfbbak9dj0zw6j1/AACxBLjCqQkvMpuz9tk0qIKMa?dl=0

ma1co, 2022-04-30:
Great, thank you for the dump!

Riaandw79, 2022-05-01:
Thank you. Keep up the good work!

closed 2022-05-03 by ma1co