ma1co/OpenMemories-Tweak #445: Nex 5R Where to Switch between pal and NTSC?

Liriel-666, 2022-08-28:
Where on the Nex 5R to Switch between Pal and NTSC Video Setting??

OpenMemories Tweak is installed and enable Pal/NTSC Selector is enable.

RenFontes, 2023-08-02:
I don't think it is possible in the 5R at least not with the current settings enabled in the app. There may be a way to do it but it hasn't been done yet.

theantnest, 2025-01-19:
Just to clarify this, it does work but, confusingly, you have to DESELECT the PAL NTSC selector option.

Also you must reformat your SD card as you cannot mix PAL and NTSC on the same card.

Liriel-666, 2025-01-19:
> Just to clarify this, it does work but, confusingly, you have to DESELECT the PAL NTSC selector option.
> 
> Also you must reformat your SD card as you cannot mix PAL and NTSC on the same card.

i can not confirm it.
It may be displayed on the camera screen 60i/p but when you open the file on a pc it says 50fps.
even with deselect the selector and format the card

theantnest, 2025-01-20:
Yes I just checked the actual files and you are correct, it still records at 50fps even though it says 60. Dang