ma1co/OpenMemories-Tweak #461: Purchased a7iii that is region locked and does not support apps

Helpmypoorass69, 2023-02-22:
hi im  new to all of this and my dumbass bought a use a7iii that it language locked to indonesian languages 
how do i get around this i have learned from reading and videos that the a7iii does not support apps but i read in another issue about drivers and what not but idk what the hell any of that meant. could someone pleaseeee help! i just want my camera in English i cant even change settings for these downloads.

Sujygunsmoke, 2023-04-06:
> xin chào, tôi mới làm quen với tất cả những thứ này và thằng ngốc của tôi đã mua một chiếc a7iii để sử dụng nhưng ngôn ngữ của nó bị khóa đối với các ngôn ngữ tiếng Indonesia. Làm cách nào để tôi khắc phục điều này. Tôi đã học được từ việc đọc và các video rằng a7iii không hỗ trợ các ứng dụng nhưng tôi đã đọc trong một vấn đề khác về trình điều khiển và những gì không nhưng idk cái quái gì trong số đó có nghĩa là gì. ai đó có thể vui lòng giúp đỡ! tôi chỉ muốn máy ảnh của mình bằng tiếng Anh, tôi thậm chí không thể thay đổi cài đặt cho các bản tải xuống này.

Liên hệ sujygunsmoke@gmail.com , tôi sẽ khắc phục .

Sujygunsmoke, 2023-04-06:
> hi im new to all of this and my dumbass bought a use a7iii that it language locked to indonesian languages how do i get around this i have learned from reading and videos that the a7iii does not [...]

Contact sujygunsmoke@gmail.com , I will fix it .