ma1co/OpenMemories-Tweak #495: DSC-WX5

ookato, 2024-06-30:
Thank you for your wonderful software.

Can you please help me with my recently acquired DSC-WX5?
It looks like the DSC-WX5 is supported, but it actually says "No tweaks available".
I read the following article and prepared a dumped firmware.

https://github.com/ma1co/OpenMemories-Tweak/issues/309

I have uploaded the dumped firmware here

https://www.dropbox.com/scl/fi/231tyj0qbu63xcpdglu5q/DSC-WX5.zip?rlkey=srix4lycb916ov72amhhlw3d5&dl=0

Can you please help me? Thank you in advance.

Translated with DeepL.com (free version)

Sujygunsmoke, 2024-09-05:
contact my email, i can fix it
sujygunsmoke@gmail.com

keranggabut, 2024-09-12:
Hi @Sujygunsmoke I have problem same with @ookato 
My Sony WX5 is supported, but it actually says "No tweaks available".
I read and the following article. But still stuck.

Can you please help me? Thank youuu

Sujygunsmoke, 2024-09-12:
yes, icant help, contact my email ***@***.***