ma1co/OpenMemories-Tweak #496: Tweaking no longer works for the A6700

Nebulater, 2024-07-13:
Hi! 

With firmware update 1.03 the language unlock, as well as the whole tweaking window no longer opens after following the steps. After the drivers are changed to win-usb the camera registers its connected to Zadig (and I did the whole Sony USB step), but when I click PMCA’s service tweak it says “No tweaks available”. 

I tried multiple versions of PMCA, Zadig and now I’m thinking of downgrading my camera to 1.02 to try it again but I don’t want to risk breaking something.

I’m using a Japanese Sony Alpha 6700 and I want to unlock English. That’s it.

Sujygunsmoke, 2024-09-05:
contact my email, i can fix it
sujygunsmoke@gmail.com

liuhiasheng, 2024-10-21:
I have a beta version of PMCA here, which must be able to support 6700

liuhiasheng, 2024-11-30:
Leave your email address.

liuhiasheng, 2024-11-30:
> > 我这里有一个 PMCA 的测试版，它必须能够支持 6700
> 
> 你能分享一下吗，这对我很有帮助。非常感谢

Give me your email address.

Administator-1112, 2024-12-01:
> > > 我这里有一个 PMCA 的测试版，它必须能够支持 6700
> > 
> > 
> > 你能分享一下吗，这对我很有帮助。非常感谢
> 
> Give me your email address.

my problem:
Querying MTP device
Sony Corporation ILCE-6700 is a camera in MTP mode

Getting device info
Error: Cannot determine camera model

another problem:

Switching to app install mode
Traceback (most recent call last):
  File "pmca-gui.py", line 78, in do
  File "pmca\commands\usb.py", line 280, in installCommand
  File "pmca\usb\sony.py", line 338, in switchToAppInstaller
  File "pmca\usb\sony.py", line 276, in _sendCommand
  File "pmca\usb\sony.py", line 129, in sendSonyExtCommand
  File "pmca\usb\driver\windows\wpd.py", line 167, in sendWriteCommand
  File "pmca\usb\driver\windows\wpd.py", line 139, in _send
Exception: MTP SendCommand failed: 0x-7ff8ffe1

 can you send e-mail to sjx081113@126.com？plz

liuhiasheng, 2024-12-10:
Your device is too new to support this script.

burande, 2025-03-23:
> I have a beta version of PMCA here, which must be able to support 6700

Hi sir, Pls 2998363987@qq.com, thank you very much

Uzver123, 2025-04-24:
Is it even meant to work on newer cameras that dont have PlayMemories Camera Apps (PMCA)? 

I doubt it ever worked for A6100, A6400, A6600 and A6700 on any firmware.