ma1co/OpenMemories-Tweak #497: sony a7s2 can not install the app

nimblefelow, 2024-07-13:
it barelly can see my device and when i manage to make it seeing the camera  it says this when I try to install an app:
Sony Corporation ILCE-7SM2 is a camera in app install mode

Downloading apk
Analyzing apk

Warning: Invalid apk file

Starting task
Starting communication
Traceback (most recent call last):
  File "pmca-gui.py", line 75, in do
  File "pmca\commands\usb.py", line 300, in installCommand
  File "pmca\commands\usb.py", line 71, in installApp
  File "pmca\installer\__init__.py", line 84, in install
ConnectionRefusedError: [WinError 10061]
maybe it's the 3.01 v of firmware but Idk, also I can't shoot 4k and 100 fps on this camera with a right sd card and also cant watch playback, maybe somebody know how to fix this

nimblefelow, 2024-10-21:
> This app requires cracked software to install

Which one, can you please help me, idk how to (

liuhiasheng, 2024-10-22:
Contact my email liuhuasheng5017@qq.com